Proto-App/Proto-Android #341: Problems with high frequencies/multiplexing

Sven130889, 2024-05-08:
I'm trying to display letters and symbols on the LED matrix using multiplexing. As a first test, I tried to make the 4 LEDs in the corners light up one after the other like a running light. When I then set the whole thing to a high frequency to create the effect that all 4 light up at the same time, the program unfortunately didn't cope properly. Maybe my tablet isn't powerful enough. Does anyone have any other ideas on how I can do this?
[image]

Proto-App, 2024-05-11:
Hello @Sven130889 
Could you share your circuit? It might be performance issue because matrix led requires strong computational power (64 LEDs to simulate)

Sven130889, 2024-05-11:
[https://www.protosimulator.com/share?circuit=Circuit_1_1715451734101](url)

Proto-App, 2024-05-12:
Thank you, we will check this case.