Proto-App/Proto-Android #345: Proto in virtualized environment

krotow, 2024-05-12:
Sorry to bother you in this matter again. However I repeat that **working with larger circuits without trace autorouting in small phone screen is self-torturing nightmare**. Because of that tried to run Proto under various virtualized desktop environments like BlueStacks, Genymotion or Waydroid. Proto works in both, however my main working OS is Linux (particularly Linux Mint 21.3) that I have no intention to move elsewhere due to work project specifics and other reasons. BlueStacks is ad-ridden thing that don't work in Linux. Genymotion is shady SaaS thing that will send customer Google account and other private info God knows where. Waydroid currently has no pinch zoom emulation (Proto is impossible to work without it!), and installing is tricky with steep learning curve (many your customers will not manage to launch it, believe me).

There is another way - Android Studio AVDs where is possible to create exact environment required for Proto together with Google Play service on both 3 main platforms ()Linux, Mac, Windows). However Proto when installed in AVD though Google Play, crash shortly after launch and refuse to run. Can you look into it? I tried both Android 11 (30) and Android 14 (34) varieties. Would be nice if it would be run Proto in AVD, especially in large resolution. I have 1440p and 1080p monitors, Proto on 13" AVD with 2000+ x 1080+ resolution would be already usable.

Proto-App, 2024-05-12:
Hello @krotow 
Yes, PROTO won't work on AVD because we have internal logic to prevent it (because of copycats etc). We always recommend to use devices like tablets. You can even use tool called scrcpy to control everything on your PC, it is very responsive.

Proto-App, 2024-05-12:
By the way we don't have any plans to unlock AVD because Android is super unsecure when root is available ;/ and it is very easy to stole our work...

krotow, 2024-05-12:
So we are back to square one - continuing to be masochist or... maybe a chance to see Proto desktop app some day? For example EveryCircuit use Chrome app which works on both platforms and seems fast enough. Can't run Proto in Waydroid - in current state it not only lacks proper screen zooming, but also steal all GUI windows from porograms, launched after Weston that is even worse.

FYI: I removed Genymotion from request because dropped a ball and tried it IRL. And of course Proto didn't worked in it like in AVD. Same copy protection, I guess.