Proto-App/Proto-Android #350: IC 7485 is maybe not working as it should be

Amonia69420, 2024-06-11:
So, i was testing the IC 7485 (4-bit magnitude comparator), and i got a few issues

So, the IC 7485 should set A=B pin to high, if A = B. But, the problem is, IC 7485 sets A>B pin and A<B pin to high, and A=B pin is still low. I don't know if is this a bug, or a feature.

Gallery:
[image]
image 1.1: A bug in IC 7485

[image]
image 1.2: A value of A and B

[image]
image 1.3: Full schematic

[image]
image 1.4: A = B (A is 0, and B is 0) represented in A>B pin high and A<B pin high, but A=B still low

[image]
image 1.5: How to fix the bug by using AND gate to A>B pin and A<B pin

Proto-App, 2024-06-12:
Hello @Amonia69420 
Thank you for report. We will check it before next release.

Amonia69420, 2024-06-13:
thanks, @Proto-App

Amonia69420, 2025-02-17:
After i read the datasheet, the IC is working properly. Sorry for inconvenience, @Proto-App

closed 2025-02-17 by Amonia69420