Proto-App/Proto-Android #352: The application closes when I try to add AD633 component

diegoxze35, 2024-06-16:
In any project's application, when I try to add AD633 component the application closes.
And when the application is reopen, the component is not in the work area.

Mi device: Galaxy A01 with Android 11.

Proto-App, 2024-06-18:
Thank you for report. We will check this case.

vinhk7agg, 2024-07-13:
> Thank you for report. We will check this case.

vinhk7agg, 2024-07-13:
> In any project's application, when I try to add AD633 component the application closes.
> And when the application is reopen, the component is not in the work area.
> 
> Mi device: Galaxy A01 with Android 11.
>

vinhk7agg, 2024-07-13:
> Thank you for report. We will check this case.

- f9ccfe088dc473e8fd162103bab5bff91b09ef1b

Proto-App, 2024-07-16:
Hello @diegoxze35 
Do you have issues with AD633?

closed 2024-09-02 by Proto-App