Proto-App/Proto-Android #354: a short question

Jurgen2003, 2024-07-05:
dear @Proto-App 
 is it possible to determine in javascript the min , max over 2 inputs for example input 1 has 5 volts and input 2 has 7.5 volts that output 1 then sends out 7.5 but becomes input 1 10 volts output 1 10 volts so over 2 inputs the highest send out   
  if so how does it work I am trying it does not work   
greeting Jurgen

Proto-App, 2024-07-11:
Hello @Jurgen2003 
Sure, you can use Math.max function:
` let maxVoltage  = Math.max(in0.getVoltage(), in1.getVoltage()); `

Proto-App, 2024-07-11:
@Jurgen2003 
Can we close this topic?

Jurgen2003, 2024-07-12:
the minumale is then let minVoltage = Math.min(in0.getVoltage(), in1.getVoltage()); 
   thanks and can be closed

Proto-App, 2024-07-12:
Yes, Math.min for minimum value.

closed 2024-07-12 by Proto-App