Proto-App/Proto-Android #355: I need help. I want to use an IC with TDA2822m specifications in the application.

1866nazari, 2024-07-14:
Hello to the support of proto application developers.
I need help.
I want to use an IC with TDA2822m specifications in the application.
Please guide me how to do it.
Thankful.
[Datasheet TDA2822M.pdf](https://github.com/user-attachments/files/16215723/Datasheet.TDA2822M.pdf)

Proto-App, 2024-07-16:
Hello @1866nazari 
We think you can create java script code and simulate behavior of this IC

1866nazari, 2024-07-17:
Hello.
thanks for your answer.
IC2822M was able to proto run.
The datasheet has been sent to you.
Excellent proto.
[image]

🙏

1866nazari, 2024-07-17:
Hello.
thanks for your answer.
IC2822M was able to proto run.
The datasheet has been sent to you.
Excellent proto.