Proto-App/Proto-Android #364: a question and sugestion

Jurgen2003, 2024-08-28:
dear proto 
i wonder if there are javascript ic coming up with more in/out corridors 
and that they are rotatable or a data contact where you can send multiple ic with 1 script i know what the possibilities are but would be nice if there is an option with which more than 8 in/out passages can work on a script 
 Yes I know I already asked but I am curious if there is an idea for it would be a nice option in addition to that an igbt would also be a nice option.

Proto-App, 2024-09-02:
Hello @Jurgen2003 
Sure, we can introduce JS integrated circuit with more inputs and outputs. How much do you need?

Jurgen2003, 2024-09-02:
I was thinking of doubling through the moves so 
 1,2,4,8,16,32 think with 32 is always enough and would ic the side input and output is interchangeable just like. Vscs power supply 
  I hope it works with 16 I would already be happy 32 offers more possibilities I would love to see a reaction jurgen

Thanks in advance

Jurgen2003, 2024-09-18:
Dear @Proto-App 
what are the possibilities and julie planning for more inputs and outputs on the javascript ic. I.gave an idea how do you guys think about it what is possible I would be happy with it 
Greetings jurgen

Proto-App, 2024-09-18:
Hello @Jurgen2003 
We will add new Script IC with more inputs and outputs in next release :)

Jurgen2003, 2024-10-09:
Dear @Proto-App 
 I am curious about the next version for my next project I want to do more with java ic 
  I wonder how many inputs and outputs the next version has so I can already write a script in preparation  
 I am waiting to build on that next version  
Greets jurgen

Jurgen2003, 2024-11-04:
dear @Proto-App proto 
thanks for the 16 in /out ic the shape is handy and looks nice i am happy with it  
 the question can be closed  
 greetings jurgen

Proto-App, 2024-11-04:
@Jurgen2003 Great👍

closed 2024-11-04 by Proto-App