Proto-App/Proto-Android #368: Feature request

GigabyteCode, 2024-09-22:
Hello you have a neat little app and I like it very much but i would like to request a feature. In the past I have used a lookalike application called 'Logic Circuit Simulator'  and it has a feature where you can set up your own IC without using code. I'm not sure how to exactly explain it but basically you make a circuit you want then you map out the IC pins and assign them to different connections on your circut and then you can import that IC into any project you want. I have added an image for better explanation. 
[image]
I think that this would be a wonderful addition to the app.

GigabyteCode, 2024-09-22:
Another feature I just thought of was when designing circuits and when the wires overlap they kinda like jump. For example:
[image]

Proto-App, 2024-10-02:
Hello @GigabyteCode 
Thank you for suggestions. We think you mean feature called "subcircuits" and it is on our TODO list. Second request with wire overlaps is already in development pipeline.