Proto-App/Proto-Android #378: Missing components

Dvante, 2024-10-23:
Some components like the 4-bit adder 74283 are missing, i would like to know if you can add them.

Proto-App, 2024-10-23:
Hello @Dvante 
We will add 7483 in future releases. If you need this IC now then you can use JavaScript addon and write code to simulate 7483 behavior.

Act77, 2024-12-27:
Hello, I like using the PROTO app very much.
Unfortunately, I am missing a component that is quite important to me.

It would be great if you could add the following component:

Voltage regulator 1117 (1.2 V , 1.5 V , 1.8 V , 2.5 V , 3.3 V , 5 V).

There are a wide variety of drawings.
AMS1117 ; LM1117 ; LD1117 .......

Proto-App, 2025-12-18:
@Dvante 
 Implemented in version 1.45.0 (will be available in next 3 days)