Proto-App/Proto-Android #383: Improvement request

TOM-FARMER-MACHINERY-SERVICES, 2024-11-13:
HI,
This is my favourite app for diagrams. there's just a few things which I'd love to see in there.
1. I'd love to be able to use longer circuit names and have the capability to create folders too.
2. I have the Java add on, however I tend to write in  C++. I understand this may be a big ask, but it would be cool to have the capability to code in this language too.
3. Moving items is difficult as you have to move them individually, and wires don't seem to be able to be moved. I'd love that in the app too.

It's a great app, I look forward to seeing any changes!

Proto-App, 2024-11-20:
Hello @TOM-FARMER-MACHINERY-SERVICES 
1. How much characters do you need? :) We will think about folders too.
2. Sorry but JavaScript can be executed in runtime on Android. For C we need specific compilers so it is much harder to compile and execute C code ;/
3. We are preparing this functionality.

Thank you for suggestions!

TOM-FARMER-MACHINERY-SERVICES, 2024-11-20:
Hi Proto,

Thank you for getting back to me!
1. I should think 30 characters should be plenty. No probs about folders, not a necessity, just a bit tidier!
2. No problem, java will be fine. The code I need to write is simple anyway, so I can either write in java, or just ommit this component for now.
3. Sounds great. I look forward to it!

I'm sorry to add more work, but I was wondering if once circuits are shorted or destroyed, they could be reset at all to restart the simulation? It seems as though it stops simulation functionality once the circuit has been destroyed, which of course is reality, but it would be cool to have the ability to repair and continue!

I really appreciate your work to improve the app and get back to me for potential improvement! Keep up the great work.

Regards,
Tom Farmer Machinery Services
07878856510
www.tfms.uk