Proto-App/Proto-Android #394: ISSUE AFTER RECENT APP UPDATE

Sourav16ee, 2024-12-11:
I have bought the full bundle pack of components a long ago and have used them even today...but as I updated the app , all the bundle pack components are not visible any more...only the unpaid version components are available...please let me know how can I fix this problem..looking forward for your kind cooperation..thanks a lot in advance..

Reshot14, 2024-12-12:
I have the same problems, my purchased components have disappeared: (

Proto-App, 2024-12-12:
Hello @Sourav16ee @Reshot14 
Could you uninstall PROTO and install once again? We will check this issue.

Sourav16ee, 2024-12-12:
@Proto-App Hello ! I just now uninstalled and then re-installed the app..no change..the same scenario persists...but now the workspace has also got completely deleted...Idk how to recover it...please advise me the way to fix the above mentioned problems..thanks a lot for your cooperation..

Proto-App, 2024-12-13:
@Sourav16ee 
Pleas share your orderID, we will check addons status on our server.

Sourav16ee, 2024-12-13:
@Proto-App
OrderID : GPA.3386-4672-8577-96052

Proto-App, 2024-12-13:
@Sourav16ee 
Strange because we can't see any request from your device to our server (in last 7 days), could you clear app data from System settings and run PROTO after that?

Sourav16ee, 2024-12-13:
@Proto-App Proto-App
Just tried clearing the app data from settings but still no change...
The purchase of the bundle pack was done on 13th October , 2023 .

Proto-App, 2024-12-17:
@Sourav16ee 
We can refund this payment, later on you can try to buy it again. Now we can't find any reason behind this issue.....

Manuel-Delsis, 2024-12-19:
Hello, I also have the same problem, in my case it only happens when I have more than one Google account on the phone, I explain I have an account where I bought PROTO, and I have another one from work, when I add the work account to my device, my purchase disappears and when I remove it it comes back

Proto-App, 2024-12-20:
@Manuel-Delsis 
Account from work is your default one? All purchases are provided from Google framework and we don't have any control over it. Maybe there is an option to set default account?

Manuel-Delsis, 2024-12-20:
As far as I know, no, it is not my main account. Before commenting here, I looked for someone who had a similar problem to mine on Google support and I came across this post from a few years ago: https://support.google.com/googleplay/thread/3113460?hl=es. (This one is in Spanish)

And the response this person received was to contact the app developer.

Proto-App, 2024-12-24:
Sure, we will revoke access to add-ons and you will get full amount soon.

Proto-App, 2024-12-24:
It will be reverted by Google :) It is in progress. 
[image]

Sourav16ee, 2024-12-24:
@Proto-App 
Thank you so much....

closed 2024-12-26 by Proto-App