Proto-App/Proto-Android #395: Purchases issue

Tomomoto10, 2024-12-13:
Hey not sure if its the update (1.35.0) is broken or if my bundle purchase is broken. I updated to the lastest update and purchased the bundle add on pack on the same day. But the purchased IC Script components and examples crash the app. Sometimes reloads proto menu but usually crashes out to device home home. ive delete the app, deleted data, reinstalled but still crashes every time. 
Error report attached 
[Error in P R O T O 9bc60fb12bb0.txt](https://github.com/user-attachments/files/18131227/Error.in.P.R.O.T.O.9bc60fb12bb0.txt)

Proto-App, 2024-12-13:
Hello @Tomomoto10 
Thank you for report. Could you share example name that crashes on your device?

Proto-App, 2024-12-13:
@Tomomoto10 And please share your order ID.

Tomomoto10, 2024-12-13:
It happens with all three IC Script examples, and all of the IC Script parts. As soon as you add the component the app flashes a white screen then kicks back out to to app menu, or it crashes completely and gives an error popup on android. 
Can do, do you mean the Order ID on the play store?

Proto-App, 2024-12-13:
Yes, please share also phone model, it looks like it is connected only with JavaScript addon. We had crashes on some devices.

Tomomoto10, 2024-12-13:
Device is: Pixel 7 Pro - GrapheneOS operating system.
Google Play Order No: GPA.3336-3486-9828-80502

From what I can find, that is the only issue. Ive added most parts and everything else seems to work no worries.

Tomomoto10, 2024-12-14:
I might be wrong but it looks like its crashing on loading the device com. libraries? Will there be a fix? No rush as im not actually using the JS right now, but would like to in the future.

Tomomoto10, 2024-12-14:
I have found the issue, it is a GraphineOS specifc issue. 
There is an "exploit protection compatibility mode" in the app settings, graphine os restrics some of the call actions for some apps due to security reasons. Ive turned this on to bypass it and now the Issues have gone.

Proto-App, 2024-12-15:
Ok, great that it works as expected.

closed 2024-12-15 by Proto-App