Proto-App/Proto-Android #399: java

Jurgen2003, 2025-02-11:
dear @Proto-App 
can the java ic measure more than just voltage for example the hertz on an input to gnd and can put the hertz on the output for example 5 volts and 10 hertz as input or output 
  and can the average voltage of an ac voltage on the input be measured yes a java can do anything you program it to do but what are the options that julie support?

Jurgen2003, 2025-02-21:
dear @Proto-App 
how does the new log function work 
greets jurgen

Proto-App, 2025-02-21:
@Jurgen2003 
In case of frequency, you can measure voltage level at the input and measure time between changes. Then you can calculate time diff between changes and you will get frequency value. Furthermore you can set output voltage at exact time and create voltage source with desired frequency.

Proto-App, 2025-02-21:
Please check following screenshots how to use logs:

[image]

[image]

[image]

closed 2025-02-23 by Proto-App