Proto-App/Proto-Android #400: one question

Jurgen2003, 2025-02-26:
dear @Proto-App proto  
here is an example of a simple script  
 the goal is at the setupTerminals a calculation is done the result must be on an output and the function driveOutputs must react on that result i found 2 options or as here use 1 output put the result on there and with the function driveOutputs read out that output but that costs an output the other way is to recalculate the sum in the function driveOutputs but if there is fixed value for example let max used in the calculation and that value is changed it will have to be changed in the function driveOutputs also   

 Now the question    
is it possible to make the function driveOutputs look at the scipt of setupTerminals   
  or are they 2 scripts separate    
 
 further i do not understand the log i did julie tip but where do you see what is logged or what can you do with the log function   
greetings jurgen 

 

function setupTerminals() {
     let sum=(in0.getVoltage()+in1.getVoltage())
     let max=(20.0) 
    out0.setLowVoltage(0.0);
    out0.setHighVoltage(sum);
    out1.setLowVoltage(0.0);
    out1.setHighVoltage(sum);
    out2.setLowVoltage(0.0);
    out2.setHighVoltage(2.5);
     out3.setLowVoltage(0.0);
    out3.setHighVoltage(5.0);
}

function driveOutputs() {
    out0.setState(true);
    out1.setState(out0.getVoltage() > 2.5);
    out3.setState(out0.getVoltage()>5.0);
    out4.setState(out0.getVoltage() >10); 
}

setupTerminals();
driveOutputs();

Translated with DeepL.com (free version)

Proto-App, 2025-02-27:
Hello @Jurgen2003 
You don't need to use setupTerminals and driveOutputs. You can create your function and join code from setupTerminals and driveOutputs like below:

```js
function setupTerminalsAndDriveOutputs() {
    let sum=(in0.getVoltage()+in1.getVoltage())
    let max=(20.0)
    out0.setLowVoltage(0.0);
    out0.setHighVoltage(sum);
    out1.setLowVoltage(0.0);
    out1.setHighVoltage(sum);
    out2.setLowVoltage(0.0);
    out2.setHighVoltage(2.5);
    out3.setLowVoltage(0.0);
    out3.setHighVoltage(5.0);
    
    out0.setState(true);
    out1.setState(out0.getVoltage() > 2.5);
    out3.setState(out0.getVoltage()>5.0);
    out4.setState(out0.getVoltage() >10);
}

setupTerminalsAndDriveOutputs();
```

Jurgen2003, 2025-03-01:
Thanks     @Proto-App

closed 2025-03-02 by Proto-App