Proto-App/Proto-Android #404: scripted IC crashes when run

joecar7, 2025-04-05:
When the run button is pressed, the scripted IC devices cause Proto to crash.
Same with the reset/refresh button.

I am running on Android 7.

Thanks

joecar7, 2025-04-08:
> Hello [@joecar7](https://github.com/joecar7) Could you share your device name?

LGMP450

( Android 7 cell phone made by LG )

Proto-App, 2025-04-09:
Do you mean this device: https://www.lg.com/us/cell-phones/lg-MP450-metro-by-t-mobile-stylo-3 ?

joecar7, 2025-04-09:
Yes, that one.

Thanks.

Proto-App, 2025-04-13:
Hello @joecar7 
We will try to add your device to different java script mode (in next release). It should solve your issue, if not then we will refund your payment.  Next release should be available within 7 days. 

Best Regards

joecar7, 2025-04-13:
> Hello [@joecar7](https://github.com/joecar7) We will try to add your device to different java script mode (in next release). It should solve your issue, if not then we will refund your payment. Next release should be available within 7 days.
> 
> Best Regards

Thanks!!!

joecar7, 2025-05-04:
I see that this is now fixed with this update, Thanks!

Also, with this update, this version of JS compiler treats the keyword "let" as syntax error. But this is not a problem, "let" is not very useful anyway.

Proto-App, 2025-05-04:
@joecar7 Great!
Yes, now your device uses second Java Script engine and there 'let' is not supported ;/