Proto-App/Proto-Android #407: question on ic's

Jurgen2003, 2025-04-28:
dear @Proto-App   
would it be possible to enlarge the javascript is to 32 input and output in addition to the 16 to add a 32 
would like to see julie response to this 
greetings
jurgen

joecar7, 2025-04-30:
Could make user definable **N** inputs/outputs, not just 16 or 32...?

Proto-App, 2025-05-04:
Yes, it might be possible in future to set N inputs/outputs. We have to rethink area detection of our components because you can expand it to much bigger size than before.

joecar7, 2025-05-05:
Or could have N inputs and M outputs (i.e. N and M can be different or same).

Jurgen2003, 2025-05-06:
dear @Proto-App proto 
I understand that space is limited but that is why it is convenient to program the circuits as much as possible is java ic so one is in smaller than a whole circuit reinforcements in an ic saves space with opams but the scipts are getting bigger that is not a problem only from an analog calculation by there the next one must now be done by an output that span to the next ic so is the inputs expand end with more ic but that costs a lot in and out puts for cominication and more space 
 hence the question for more inputs and outputs to control 
 thought at 16 is enough but those are already full too 
 reducing the components is an option but the user can still consider for himself how he handles the ic 
 and inder daad if you can add them yourself you can use it all by yourself the big ic become more processor than an ic but that is not a problem 
 looks nice too
 
 furthermore there is a question 
 programming an in and out channel is no problem for example < 2.5 and below that 
 but is it possible to program an out channel that works as a shadow trigger 
 so at 5v on and below 2.5 off only then the voltage in between is bridged

greetings jurgen

Jurgen2003, 2026-10-03:
Dear @ Proto @joecar7 
, I’m wondering how far you’ve got with the larger JavaScript IC for 24 or 32 – preferably 32 – a protocol that allows the Scpits to work together. I’m currently working on a protocol to send the values over a connection, as a sort of network, but could you perhaps programme something that works within the app for communication using JavaScript?  is getting so cluttered? Could you come up with an option for more inputs and outputs? My idea for the network is intended to have one script with others acting as input/output ports, like the I/O boards on a computer

Proto-App, 2026-10-07:
Hello @Jurgen2003 
JavaScript with 32 in and out will be delivered in next release :) But it will require quite decent phone because JS computation might be quite heavy for 64 pins.

Proto-App, 2026-10-08:
@Jurgen2003 
IC is in test phase:

[image]

Jurgen2003, 2026-10-08:
@Proto-App 
 It’s  good to see  that you’re responding to questions  and suggestions;  I’m pleased that the 32 IC is on its way   
 And thank you  for the  response  from  you all