Proto-App/Proto-Android #408: ENHANCEMENT/FEATURE

Sourav16EE04, 2025-05-03:
Hello , 
I have made some interesting circuits . Is it possible to make them publicly usable ready made components . Also can there be a feature in the app in the future by which we can enclose a circuit as a component that can be used in other circuits ? 
Please let me know if you want to have a look at those circuits so that you can evaluate if they worthy of making them as publicly usable components . 
Looking forward to your kind cooperation .

Proto-App, 2025-05-04:
Hello
In case of public circuits, we have feature called "Online library", it is done in 80%. We would like to release it after summer. 
We have plans to introduce feature called subcircuits and add option to create enclose own components as one IC. Currently XY mode is not in our development phase.

Sourav16EE04, 2025-05-04:
Thanks a lot . Really excited for those new features . Great work PROTO Team !