Proto-App/Proto-Android #412: 407  clossed

Jurgen2003, 2025-05-23:
dear @Proto-App  
I understand that space is limited but that is why it is convenient to program the circuits as much as possible is java ic so one is in smaller than a whole circuit reinforcements in an ic saves space with opams but the scipts are getting bigger that is not a problem only from an analog calculation by there the next one must now be done by an output that span to the next ic so is the inputs expand end with more ic but that costs a lot in and out puts for cominication and more space
hence the question for more inputs and outputs to control
thought at 16 is enough but those are already full too
reducing the components is an option but the user can still consider for himself how he handles the ic
and inder daad if you can add them yourself you can use it all by yourself the big ic become more processor than an ic but that is not a problem
looks nice too

furthermore there is a question
programming an in and out channel is no problem for example < 2.5 and below that
but is it possible to program an out channel that works as a shadow trigger
so at 5v on and below 2.5 off only then the voltage in between is bridged

greetings jurgen

Proto-App, 2025-06-12:
Hello @Jurgen2003 
Yes, you can write code that simulates described behavior. Please share your Google email address, we will add you as beta tester :)

Jurgen2003, 2025-06-12:
hallo @Proto-App 
my email is jurgenstorm@gmail.com 
furthermore could you guys make a preview for 1 output with that behavior then I can integrate it into my script
groet jurgen

Proto-App, 2025-06-12:
@Jurgen2003 Sure, we can prepare this code. Do you want schmiitt trigger bahavior: https://en.wikipedia.org/wiki/Schmitt_trigger ?

Jurgen2003, 2025-06-13:
yes that behaviour

Jurgen2003, 2025-07-06:
Dear @Proto-App 
How do you program a schimmit trigger in javascript and would a 24 in and out going ic be possible I now divide it over 2 ic of 16 but is not easy 
 Yes I know it is the same as the previous one but had not had any response hence this post 
 Greets jurgen

Proto-App, 2025-07-06:
Hello @Jurgen2003 
Sorry fore delay but we had holiday season. We changed beta testers bucket size, please check available beta test link tomorrow on Google Play page.

Proto-App, 2025-07-28:
Hello @Jurgen2003 
Did you join our beta test path on Google Play?

Jurgen2003, 2025-07-28:
Yes I will but is still full