Proto-App/Proto-Android #413: Suggestions for futures updates

nath654nath, 2025-06-21:
Hey, I love PROTO it's a beautiful app. I'm using it since around one year and I have some suggestions for your futures versions of PROTO 

- Arduino support (In progress I saw it on google play) 
- be able to move components in groups
- be able to copy and paste them
- add programmable ICs in python
- add a clock component ( there is only a pulse component, not a clock one) 
- add a PWM component
- infinite build space 

Thanks for this beautiful app! Have a good day!

Proto-App, 2025-06-27:
Hello @nath654nath 
Thank you for feedback! Could you share more info about clock component and PWM? Maybe share some example?

nath654nath, 2025-06-27:
Bonsoir ! Bien sûr !!

Alors, pour l'horloge ce serait dans la logique un simple signal de clock.
Il serait modulable comme l'est par exemple une résistance. Il peut être
par exemple modulable de. 1pHz à 10 GHz. Cela permettrait d'avoir un signal
d'horloge simple et régulier, pouvoir faire des circuits plus compactes et
éviter les grosses installations comme le 555.

Pour le modulateur PWM. Il m'aurait été très pratique lors de la simulation
d'un de mes dernier circuits. Il m'a été malheureusement impossible d'en
faire un stable avec le 555. On pourrait le régler manuellement comme le
clock.

Peut-être qu'on pourrait le programmer? Avoir quelques entrées et permettre
(en python par exemple) de dire que si IN0 est à 1, le signal de sortie
qu'on a programmé à par exemple PWM127 s'activerait, etc. Il pourrait avoir
5 entrées par exemple. Et 3 d'entre elles pourraient être analogique. Si
c'est 0V en entrée, ça sera PWM000 en sortie, si c'est 2.5V ce sera PWM127
et si c'est 5V ce sera PWM255.

Aussi, je trouverais ABSOLUMENT INCROYABLE d'avoir une zone de construction
INFINIE. Parce que c'est très contraignant d'avoir un si petit espace de
construction quand on aime faire de gros circuits 😅

Ensuite, qu'en pensez-vous du microcontrôleur programmable en python?

Cordialement,
Nathanaël.

ah-git, 2025-07-05:
Grouping would be very useful. Additionally if groups could optionally be represented as an IC with input and output pins.

Jmikesell90, 2025-08-01:
I would love to see Landscape supported in future versions. I use a tablet w/ keyboard and having to remove my keyboard everytime gets cumbersome and it would be better for sharing with my classroom.