Proto-App/Proto-Android #424: Beta testers - Google Play

Proto-App, 2025-08-08:
Topic for any feedback and issues related to beta version of the PROTO application.

Proto-App, 2025-08-08:
Hello @Jurgen2003 
Version 1.40.0 is available in beta channel. If you have any issues/suggestions please let us know. Thank you

Jurgen2003, 2025-09-29:
Dear @Proto-App 
The set voltage is now reduced from the resistance, but there is still
Thousands of voltages are needed to lower the resistance can there be a setting so that if you set 100 volts, at that 100 volts the resistance is 1 milli-ohm or something in that direction so that those high voltages are not needed when setting the voltages, it would be best
Greets

Proto-App, 2026-01-01:
Hello @Bellbirdbits 
No, you will get only regular update.