Proto-App/Proto-Android #426: No tengo el complemento de analog lo d

telcom1951-hub, 2025-08-15:
Pagué por el complemento y me funcionó dos días entonces????

Proto-App, 2025-08-16:
Hola @telcom1951-hub 
Desinstala PROTO y vuelve a instalarlo. Debería funcionar.

telcom1951-hub, 2025-08-17:
Buenas noches lo desinstalé y volví a instalar y sigo sin tener acceso al
complemento de analogos

Proto-App, 2025-08-17:
Hola, consulta el sitio web de Google: https://support.google.com/googleplay/answer/1050566?hl=es 

Si no funciona, te reembolsaremos tu pago.

telcom1951-hub, 2025-08-17:
Good afternoon

I'm not interested in a refund. I want the app to work.

That's why I'm contacting you, the developers.

Why am I trying to pay again and it still won't work?

Gustavo

------------------------------
Comentarios
Practica hablar en inglés
Práctica
<https://www.google.com/url?q=https://speak.google.com/speakingpractice/activity%3FnativeLang%3Des-419%26p%3DEhIKAmVzEgZlcy00MTlAAUgCWAEgAQ%26idToken%3DAf0rSnb9qDDb2Fn3VWzQ3m2ifhvWX9wLk1K7KoaeuHxkvK1vJdqvGr2Kt4wLvoPr6wVf182Y&sa=U&sqi=2&ved=2ahUKEwifj5ySwJKPAxX8TDABHQ3jAjMQ_9IKegQIBxAB&usg=AOvVaw3ur6l5N-yuWLt6B-ZarIfb>

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#m_6872592875894393693_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Proto-App, 2025-08-17:
Sure, we will try to resolve your issue. Are you able to buy it again? Or maybe you see "Owned" or "Pending" status?

Proto-App, 2025-08-25:
@telcom1951-hub 
Do you still have this issue?